bblanchon/ArduinoContinuousStepper #1: How to implement the "void" functions like powerOn, powerOff, etc. into my Arduino code?

Dicsoft, 2022-05-27:
Dear Benoît,
many thanks for your swift reply!
Your "Arduino Continuous Stepper" lib fits perfectly to my project for soft start, stop and constantly run a Nema23 stepper motor attached to a M542C driver. 
I intend to use a quite simple code with some push buttons to start, stop, change direction and increase/decrease the speed.
Finally a LCD display should show some figures but just, if the rest is running smooth and reliable.
As a processor I'm using the TEENSY LC. Actually I don't use any VOID commands and I hope that this will make my code more elegant and less timing critical because the motor don't speed up and run as smooth as it does with your example code. 
Also if I set the speed to 0, speeding down sounds not as smooth as it should be.
What I would like to understand is how I have to implement the VOID commands in my code and how to link my push buttons to it?
Please find my actual code attached...!

Many thanks for your time and support and have a nice weekend.
Dirk

`
// TEENSY LC ARM Cortex-M0+ processor at 48 MHz, 62K Flash, 8K RAM
// Steppermotor StepperOnline 23HS30-2804S 1.9Nm, 2.8A
// Stepperdriver Leadshine M542C (Dips 1on, 2off, 3on, 4,5off, 6,7,8on)
// Button unbounce with ezButton
// Continuous Stepper library for smooth ramping and continuous run
// Function enableState to toggle between Start/Stopp with SafetySlowSpeed
// Function Direction to toggle between CW/CCW with SafetyLimit 
// Button counter the easy way
// LiquidCrystalDisplay 20col/4Row with I2C

//#include <Wire.h> 
//#include <LiquidCrystal_I2C.h>
#include <ezButton.h>
#include <ContinuousStepper.h>

//LiquidCrystal_I2C lcd(0x27,20,4);  // LCD address 0x27 (Pin SDA 18, SCL 19) 20 chars, 4 line display

// Defines pins numbers

const uint8_t stepPin = 8;
const uint8_t dirPin = 7;
const uint8_t enablePin = 4;
//const int stepPin = 8;                      // Steps signal to the driver
//const int dirPin = 7;                       // Direction signal to the driver
//const int enablePin = 4;                    // Enable signal to the driver
const int enableButton = 5;                 // Enable push button
const int dirButton = 6;                    // Direction push button
const int upButton = 9;                     // Speed up push button
const int downButton = 10;                  // Speed down push button
ezButton changeDir(dirButton);              // Direction push button debounce
ezButton enableMot(enableButton);           // Enable push button debounce
ezButton speedupButton(upButton);           // Speed up push button debounce
ezButton speeddownButton(downButton);       // Speed down push button debounce
ContinuousStepper stepper(stepPin, dirPin, enablePin);

int customDelay; // Defines variables
int intervalBetweenSteps = 200;             // 200
int buttonPushCounter = 0;                  // counter for the number of button presses
int buttonStateUp = 0;                      // current state of the button Up
int lastButtonStateUp = 0;                  // previous state of the button Up
int buttonStateDown = 0;                    // current state of the button Down
int lastButtonStateDown = 0;                // previous state of the button Down
int rampVal = 1000;                           // Ramp value for Speed
int enableState = LOW;                      // the current state of the ENABLE Signal
int dirState = LOW;                         // the current state of the DIR Signal
int speedSet = 0;

 
void setup() {
 Serial.begin(9600);
  
  // Sets the pins as Outputs/Inputs
  pinMode(stepPin,OUTPUT);                   // set arduino pin to output for steps
  pinMode(dirPin,OUTPUT);                    // set arduino pin to output for direction
  pinMode(enablePin,OUTPUT);                 // set arduino pin to output for motor enable
  pinMode(enableButton, INPUT_PULLDOWN);     // set arduino pin to input pull-down mode
  pinMode(dirButton,INPUT_PULLDOWN);         // set arduino pin to input pull-down mode
  pinMode(upButton,INPUT_PULLDOWN);          // set arduino pin to input pull-down mode
  pinMode(downButton,INPUT_PULLDOWN);        // set arduino pin to input pull-down mode

  changeDir.setDebounceTime(50);
  enableMot.setDebounceTime(50);
  speedupButton.setDebounceTime(50);
  speeddownButton.setDebounceTime(50);
 
  digitalWrite(enablePin,LOW); //Disables motor

/* 
    lcd.init();                                // initialize the lcd 
    lcd.init();
    Print a static message to the LCD.
    lcd.backlight();
    lcd.setCursor(0,0);
    lcd.print("Smoothing Roller DSC");
    lcd.setCursor(0,1);
    lcd.print("Actual Speed:");
    lcd.setCursor(0,2);
    lcd.print("Direction:");
    lcd.setCursor(0,3);
    lcd.print("Status:");

  //stepper.spin(speedSet);
*/
  
}
void loop() {
  
  changeDir.loop();                           // start debounce
  enableMot.loop();                           // start debounce
  speedupButton.loop();                       // start debounce
  speeddownButton.loop();                     // start debounce
  stepper.loop();

  //speedSet = buttonPushCounter * rampVal;               // calculate the speed with pushButtonCounter * rampVal
  //stepper.spin(speedSet);

  Serial.print(F("SPEED = "));
  Serial.println(speedSet);
  Serial.print(F("OUT = "));
  Serial.println(buttonPushCounter);

  // --------------------------------------------------------------------
  // Up button counter
  // --------------------------------------------------------------------
  
  if(speedupButton.isPressed())
      buttonPushCounter++; 
     
     if (buttonPushCounter > 16)  // Max. Speed Limiter  
        buttonPushCounter = 16;
      

  // --------------------------------------------------------------------
  // Down button counter
  // --------------------------------------------------------------------
{  
   if(speeddownButton.isPressed())
      buttonPushCounter--;
   
    if (buttonPushCounter < 1)  // Min. Speed limiter   
           buttonPushCounter = 0;
      
   
}  
/*
    if ((changeDir.isPressed()) && (buttonPushCounter < 5)) {  // direction push button active at speed < 5 to protect the motor
    // toggle state of DIR
    dirState = !dirState;
    // control EnablePin arccoding to the toggled state
    digitalWrite(dirPin, dirState);                            // write signal to direction pin
    }
 */   
    if (enableMot.isPressed()) {                               // enable push button always active even for emergency stop
    // toggle state of ENABLE
    enableState = !enableState;
    // control EnablePin arccoding to the toggled state
    //  digitalWrite(enablePin, enableState);  // write signal to enable pin
     }  
    
      if (enableState == HIGH) {                              // if motor disabled, set Push button counter to min
          buttonPushCounter = 0;
  }

    speedSet = buttonPushCounter * rampVal;                   // calculate the speed with pushButtonCounter * rampVal
    stepper.spin(speedSet);
  
}

`

bblanchon, 2022-05-28:
Hi @Dicsoft,

I'm sorry, I don't understand your question.
What's wrong with `powerOn()` and `powerOff()`?

However, looking at your program, something immediately strikes me: you shouldn't call `spin()` on each loop iteration but only when you want to change the target speed, so only when a button is pressed.
That's unlikely the response to your problem, but this is how the library is supposed to be used.

Regarding the slowing down sound when the speed is set to zero, I've spent a lot of time making sure that the deceleration is as linear as possible. I compared it with the speed profile from the AccelStepper library, and I can confirm that both libraries give the same results, at least on my hardware.

Best regards,
Benoit

Dicsoft, 2022-05-29:
Good Morning Benoît,

I´m sure that your code is perfect but I’m still a noob and honestly need a tiny push/example in the right direction,
how to handle such commands like "powerOn() and powerOff()“ in my code.
Would love to have some more short examples how to implement your library.

Even what you mentioned not to call the „spin()“ function in each loop seems absolutely valid to me, but actually I don’t now how to modify my code that this function is called
just when a new target speed is available.
Sorry to bother you with that. I love tinkering with the Arduino but I missed learning C++ 30 years ago and so it’s now even harder for me.

Anyway, glad that you take your time.

Wish you a nice Sunday.

Dirk

bblanchon, 2022-05-30:
Hi Dirk,

Here is how you could call `powerOn()` and `powerOff()`:

```c++
if (enableMot.isPressed()) {
    enableState = !enableState;
    if (enableState) 
        stepper.powerOn();
    else
        stepper.powerOff();
}
```

As I told you, I do [live mentoring on Codementor](https://www.codementor.io/@bblanchon).
I often teach Arduino programming there, so don't hesitate to schedule a session.

Best regards,
Benoit

Dicsoft, 2022-05-30:
Dear Benoit,

this example is exactly what I was looking for - many thanks for that push!
Regarding an Arduino session… I already have you on my list and will come back to you
surely.
Thanks again and if you have PayPal for some kind of a donation… please let me know.
Actually I’m on a business trip and not able to try your hint… but I will asap.

Kind regards,
Dirk

Gruß Dirk

closed 2022-08-20 by bblanchon