bblanchon/ArduinoContinuousStepper #2: Is there a way to use the Arduino motor control shield, which has two dir pins, two enable pins?

KeepingOn, 2022-06-16:
Perfect for my application of having a telescope tracker - continuous motion needed for long periods. Suspect it doesn't support the driver board, buy wanted to ask.

bblanchon, 2022-06-16:
Hi @KeepingOn,

Indeed, I don't think it will work with the motor shield, but I can probably add support in a future version.
The only problem is that I don't have the hardware to test...

Best regards,
Benoit

bblanchon, 2023-08-07:
Hi @KeepingOn,

From what I read on the Arduino Motor Shield, the four pins are not "two dir pins, two enable pins" but four bridge drivers that you could use for a four-pin stepper motor or two DC motors.

Version 3.0 of the library supports four-pins stepper motors, so I think it's safe to say it is now compatible with the Arduino Motor Shield.
Note, though, that only the "full step drive" mode is supported at the moment.

Best regards,
Benoit

closed 2023-08-07 by bblanchon