bblanchon/ArduinoContinuousStepper #3: Start and Stop buttons with Accel and Decel

DavidMarrazzo, 2022-08-18:
Hey I'm new to this stuff and accelstepper was not intended for what I'm trying to do, glad you built this library but I can't figure out how to get smooth accel/decel with buttons, just wondering if you could post a simple example with just a Start button and Stop button with accel/decel for both, thanks!

bblanchon, 2022-08-20:
Hi David,

I'm sorry, but I don't have an example at hand; however, it should be fairly easy to write.
When the start button is pressed, call `spin(speed)`; when the stop button is pressed, call `stop()`.
If the acceleration isn't smooth enough, call `setAcceleration()` with a smaller value (the default is 1000).

Best regards,
Benoit

guillaume-dorczynski, 2022-10-14:
Hello,

Here is a quick example : https://wokwi.com/projects/345467092978369107

closed 2022-10-29 by bblanchon