bblanchon/ArduinoContinuousStepper #6: Need a way to invert the logic state of the Enable pin

guillaume-dorczynski, 2022-10-12:
Hello,

Please could you add an option to invert the Enable pin

My problem is with A4988 driver, the Enable pin is active LOW so using powerOn() doesn't work because it's setting the pin HIGH :)

For now, a workaround is to manually write LOW to the pin after calling powerOn()

bblanchon, 2022-10-13:
Hi Guillaume,

Please test the `setEnablePin()` it added in the ["issue6" branch](https://github.com/bblanchon/ArduinoContinuousStepper/tree/issue6).
I currently don't have the required hardware setup in place, so I could not test.
I'm waiting for your feedback before merging this branch.

Best regards,
Benoit

guillaume-dorczynski, 2022-10-13:
Merci

On line : https://github.com/bblanchon/ArduinoContinuousStepper/blob/issue6/ContinuousStepperBase.h#L42

I think it would be better to start with the Enable pin disabled, and let user call powerOn() later. Currently the motor is immediately powered on when begin() is called, which I believe is not what most people expect

bblanchon, 2022-10-14:
I'm glad to get you
r opinion on this

guillaume-dorczynski, 2022-10-15:
Sorry for yet another problem, I should have tested your changes more carefully

In previous version, in `begin()` you called `powerOn()`, which sets `_status = WAIT` to allow motor to move. But now you removed `powerOn()` from `begin()`, so `_status` is not set correctly and the motor will not move, until `powerOn()` is called by user, even after `setEnablePin()` was called or if the enable pin is not connected to the arduino (so that it's enabled by the driver board)

I made another example, it's basically the same code as my previous example, but now I included the code of your library directly so you can modify it and do live testing to solve the issue(s) : 

https://wokwi.com/projects/345591766472196690

bblanchon, 2022-10-15:
I just (forced) pushed the fix for the `_status = WAIT` and added a third parameter to `setEnablePin()` so you can control the initial state:

```c++
stepper.setEnablePin(10, LOW, false);
```

guillaume-dorczynski, 2022-10-15:
I can see the changes to `setEnablePin` (thanks!), but no changes of `_status` in `begin()` so there is still a problem if user doesn't use `setEnablePin` (for example if he didn't connect the Enable pin to the arduino)

bblanchon, 2022-10-15:
OK, this is a complete disaster.
I removed the changes from the master branch and will take the time to work on this correctly.
Sorry about that, it really is not representative of how I usually work.

guillaume-dorczynski, 2022-10-15:
I agree, need to think of a better way to do this. You can't just initialize `_status` to `WAIT` or `powerOn()` will be useless because of the condition `if (_status != OFF) return;`

bblanchon, 2022-10-29:
The new (tested) version is available on the master branch.
Note that I didn't restore the three-parameters version of `setEnablePin()`; instead, you must call `powerOff()` first.
For example:

```c++
stepper.begin(10, 11);
stepper.powerOff();
stepper.setEnablePin(12, LOW);  // set pin 12 to HIGH
```

Please give it a try.
I'll publish a new release as soon as you confirm it works.

commit bec317be (2022-10-15): Add `setEnablePin()` and allow inverting the pin's level (resolves #6)
commit ca7cd9e9 (2022-10-15): Add `setEnablePin()` and allow inverting the pin's level (resolves #6)
commit bcd97e20 (2022-10-29): Allow inverting the pin's level (resolves #6)
closed 2022-10-29 by bblanchon