bblanchon/ArduinoContinuousStepper #12: possible bug with teensyTimerTools

harrih, 2023-03-03:
Am attempting to run as high a number of steppers on a single teensy3.x  as practicable
need   around 15000PPS max
trying teensyTimerTool   example     but no success 

 step output defaults to    499.75Hz 
  regardless of requested spin speed    both positive and negative                
except  stepper.spin(0)  where no output at all    

is this a fault or incorrect usage

is there a better way to achieve objective?

TIA 
Harry Harrison

bblanchon, 2023-03-04:
Hi Harry,

I'm using TeensyTimerTool on a Teens 4.1 at 6000 steps/s, and I'm pretty sure it also worked at 10K.
However, I don't know if 15KHz is possible.
Maybe you should try PWM with [ContinuousStepper_AWF](https://github.com/bblanchon/ArduinoContinuousStepper/blob/v2.2.0/examples/AwfExample/AwfExample.ino).

Best regards,
Benoit

harrih, 2023-03-05:
Thanks
will try
Harry

closed 2023-08-05 by bblanchon