bblanchon/ArduinoContinuousStepper #14: Another problem with stop()

Rocket-Rod, 2023-08-27:
I am using the ToneExample on Arduino UNO which appears to have much smoother acceleration and deceleration over the BasicExample.
The circuit has a SPDT switch (ON-OFF-ON) which is being used to select, Left, Right or Stop.
If I only use stop(); in the else statement the stepper just continues to rotate. Only when I use spin(0); can I get the stepper to decelerate and stop. Do I have a coding issue or is there something else occurring?
[ContinuousStepper_Left_Right_Stop.zip](https://github.com/bblanchon/ArduinoContinuousStepper/files/12447414/ContinuousStepper_Left_Right_Stop.zip)

bblanchon, 2023-08-28:
Hi @Rocket-Rod,

You must call `loop()` for `stop()` to work.
Looking at your code, I can see that `loop()` it not always called; you should move it out of the `if` block.

Best regards,
Benoit

Rocket-Rod, 2023-08-31:
Hi Benoit,
Thank you for the feedback. I have moved stepper.loop() as you suggested. However, I am getting quite strange results as I will try my best to explain.
ContinuousStepper_BASIC accelerates much faster than is configured (almost immediately). With Spin(400) & setAcceleration(50) I would expect the time to reach the target speed to be approx. 8 seconds. This is not occurring in my project as can be seen in both the video & the printed timeline. Deceleration appears to be good except not as smooth as TONE.
ContinuousStepper_TONE also accelerates much faster (much the same as BASIC). Deceleration is very smooth and within the expected time except (approx. 8s) but as it comes to a complete stop there appears to be some jerking motion happening.
As I am somewhat inexperienced I am hoping you can point out what I may be doing wrong.
Kind Regards

[ContinuousStepper_Basic.zip](https://github.com/bblanchon/ArduinoContinuousStepper/files/12482730/ContinuousStepper_Basic.zip)
[ContinuousStepper_Tone.zip](https://github.com/bblanchon/ArduinoContinuousStepper/files/12482732/ContinuousStepper_Tone.zip)
[image]
[image]

https://github.com/bblanchon/ArduinoContinuousStepper/assets/105567951/41acb69c-d2a1-43f4-a708-091968dd4418

https://github.com/bblanchon/ArduinoContinuousStepper/assets/105567951/5658fb3c-a208-4d07-b76c-e717e3c1c704

bblanchon, 2023-08-31:
Thank you very much for reporting this issue with clear details, code, and even videos 👍 
I'll take more time to investigate and hopefully solve it, but here is what I can tell you after a quick glance.

I'm pretty sure the acceleration issue is caused by the fact that your program calls `spin()` continuously instead of just when it needs to change the speed.
I can probably work around this issue by measuring the elapsed time between calls to `spin()`.
Meanwhile, please try to call `spin()` only when the buttons change state, and let me know if it fixes the issue.

I suspect the issue with the Tone library is due to the [31 Hz lower limit](https://www.arduino.cc/reference/en/language/functions/advanced-io/tone/).
Can you confirm that you are using an AVR microcontroller?
I can probably work around this issue by setting a minimum speed of 31 steps/seconds.

Rocket-Rod, 2023-09-01:
Hi Benoit,
I can confirm that only calling spin() once allows acceleration to work much better. I also noted that as setAcceleration() values are increased a decline in jerky stops is observed. At setAcceleration(400) almost smooth stops occur.
I am using an Arduino UNO R3 & please correct me if I am wrong, but my limited research indicates these are in family of AVR microcontrollers.
Thank you once again for providing valuable guidance & understanding.
Kind Regards
[ContinuousStepper_Tone_V2.zip](https://github.com/bblanchon/ArduinoContinuousStepper/files/12495018/ContinuousStepper_Tone_V2.zip)

bblanchon, 2023-09-01:
Thank you for the update, @Rocket-Rod.
Arduino UNO R3 uses an AVR microcontroller, which confirms the 31Hz theory.
I'll fix these issues in a new library version in a few days.

bblanchon, 2023-09-02:
I just published version 3.0.2 of the library to address these issues.
It should be available in the Library Manager in a few hours.

closed 2023-09-02 by bblanchon