bblanchon/ArduinoContinuousStepper #15: Use the library without acceleration

kuworking, 2024-01-27:
Hello,

Is it possible to use the library without any acceleration? I can stop the nema by `stepper.powerOff();`, but I cannot start it without acceleration.

If I use high accelerations it goes fast but the stops with strange noises, so I'm stuck with using accelerations at the end, something that I'd like to eliminate.

This is my code:

```
// ----------------------------------------
// NEMA

#include <ContinuousStepper.h>

const int motor1_pdir = 2;
const int motor1_pstep = 3;
const int motor1_enable = 4;
const uint16_t potReadInterval = 1000;  // in milliseconds
int speed = 5000;
static unsigned long lastPotReadTime = 0;

ContinuousStepper<StepperDriver> stepper;

void setup() {
  Serial.begin(9600);

  // NEMA
  pinMode(motor1_pdir, OUTPUT);
  pinMode(motor1_pstep, OUTPUT);
  pinMode(motor1_enable, OUTPUT);

  stepper.begin(motor1_pstep, motor1_pdir);
  stepper.setAcceleration(1000); // higher than 1000, it stops before reaching the potReadInterval
  stepper.spin(speed);  // rotate at 200 steps per seconds
}

void loop() {
  unsigned long currentTime = millis();

  if (currentTime - lastPotReadTime >= potReadInterval) {
    speed = -speed;
    stepper.powerOff(); 
    stepper.powerOn(); 
    stepper.spin(speed);
    lastPotReadTime = currentTime;
  }

  stepper.loop();  // this function must be called as frequently as possible
}

```

So at higher accelerations it just stops after having reached the speed (or this is what I think to see), before reaching the millis interval.

bblanchon, 2024-01-28:
Hi @kuworking,

I suspect there is an integer overflow when the acceleration is high.
I'll look into it, but if you don't want the smooth acceleration, you don't need to use this library; you can use `tone()` instead.

Best regards,
Benoit

bblanchon, 2024-01-28:
@kuworking, I could not reproduce the issue with my test program, which eliminates the integer overflow theory.

Can you provide more details about your setup?
What microcontroller are you using?
What motor are you using?
Are you sure the motor supports this high speed?
Can the power supply produce enough current?
Does it work if you remove `powerOff()`/`powerOn()`?
Does it work if you remove the potentiometer handling?

kuworking, 2024-01-28:
> @kuworking, I could not reproduce the issue with my test program, which eliminates the integer overflow theory.
>
> Can you provide more details about your setup? What microcontroller are you using? [...]

I'am using a drv8825 with a cloned arduino ONE, a cheap Nema17 (can't really tell the specs).

The power supply should produce 3-5 A (I don't remember now) but it should be more than enough).

The powerOff() instruction comes long after the motor has stopped, so I don't think it makes any difference (I can test it tomorrow).

And, I don't have a potentiometer handling (?), I mean I do have it, but the code doesn't use it (have I understood you well?).

What I see is that when I use high speed, I can see (visually I mean) that the rotor moves at high speed for a few seconds, and then it stalls / stops).

Then, ideally I should be able to set the motor to accelerate at that same high speed and keep moving without stopping.
But the parameters that I find that the motor doesn't stop, give me a lower acceleration than before.
So in other words, my nema is capable of starting at that high speed. I don't care of the final speed, I just want it to accelerate as fast as possible.

So, the workaround I'm using now is to lower the speed after 50-100 ms to prevent the motor to stop, that kind of works, but my understanding is that I should not need to do that (ideally).

So, to be clear, what I'm trying to accomplish here is to have the motor to start at the fastest speed possible with no acceleration (ideally, going from 0 to max instantaneously). As fast as my nema allows it, without really caring about having a fast speed at the end. I just need to not to start slowly.

*I am a newby in arduino, I'll search for `tone()`

This is the relevant part of the code I'm using:

```c
#include <ContinuousStepper.h>

const int motor1_pdir = 2;
const int motor1_pstep = 3;
const int motor1_enable = 4;

ContinuousStepper<StepperDriver> stepper;

void setup()
{
  // NEMA
  pinMode(motor1_pdir, OUTPUT);
  pinMode(motor1_pstep, OUTPUT);
  pinMode(motor1_enable, OUTPUT);

  stepper.begin(motor1_pstep, motor1_pdir);
}

int direction = 1;
unsigned int time1 = 0;
unsigned int time2 = 0;

void motor_start()
{
  stepper.powerOn();
  digitalWrite(motor1_enable, LOW); // Enable the motor
}

void motor_stop()
{
  stepper.powerOff();
  digitalWrite(motor1_enable, HIGH); // Disable the motor, powerOff doesn't seem to prevent the motor to heat
}

void set_motor(int direction, int acceleration, int speed, int time_1, int time_2)
{
  stepper.setAcceleration(acceleration);
  stepper.spin(direction * speed);
  time1 = time_1;
  time2 = time_2;
}

bool buttonMoving = false;
unsigned int lastTime = 0;
unsigned int currentTime = 0;

void loop()
{
  handleButton(but_p1, but_1); // check if the button has been pressed
  handleButton(but_p2, but_2);

  if (but_1 == 1 && !buttonMoving)
  {
    buttonMoving = true;
    direction = 1;
    set_motor(direction, 20000, 40000, 50, 500); // acceleration, speed, 1st stop, 2nd stop in ms
    motor_start();
    lastTime = millis();
  }
  if (but_2 == 1 && !buttonMoving)
  {
    buttonMoving = true;
    direction = -1;
    set_motor(direction, 20000, 40000, 50, 500); // acceleration, speed, 1st stop, 2nd stop in ms
    motor_start();
    lastTime = millis();
  }

  if (buttonMoving)
  {
    move();
    speed = 0;
  }

  stepper.loop(); // this function must be called as frequently as possible
}

bool seen = false;
void move()
{
  currentTime = millis();

//////// >>>>>>>>> here I lower the speed, preventing the motor to stop
  if (!seen && (currentTime - lastTime >= time1))
  {
    seen = true;
    stepper.setAcceleration(1000);
    stepper.spin(direction * 2000);
  }
//////// <<<<<<<<<<<

  if (currentTime - lastTime >= time2)
  {
    motor_stop();
    seen = false;
    buttonMoving = false;
  }
}
```

bblanchon, 2024-01-29:
> I'am using a drv8825 with a cloned arduino ONE, a cheap Nema17 (can't really tell the specs).

The clock frequency is only 16MHz on an Arduino UNO, which seriously limits speed.
My measurements give a maximum speed of 3000 step/s, with nothing else running on the board.

The computation of the acceleration curves consumes most of the CPU time.
When the target speed is reached, this computation is skipped, and the UNO suddenly reaches 18000 step/s!

[image]

This could produce a brutal acceleration or freeze the motor if 18000 is too high for this motor.

There isn't much it can do about it.
Sure, I could limit the speed by performing the computation even when the speed is constant, but this sounds like a waste of CPU time.

Again, the whole point of this library is to have smooth acceleration.
If you don't need this feature, use `tone()`.

closed 2024-01-29 by kuworking